ros2/rclpy #1398: Troubling: calling `spin_until_future_complete` one one service causes another service's callback to not be called

zkytony, 2025-01-14:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22.04
- Installation type:
  binaries
- Version or commit hash:
  ROS2 Humble
- DDS implementation:
  FastDDS
- Client library (if applicable):
  rclpy

#### Steps to reproduce issue
Here I have two minimal programs. One creates a service, and the other creates a client that calls that service. The catch is that on the server side, the service is created after calling a ListParameters service on the rviz node.
```python
# minimal_buggy_server.py
import threading

import rclpy
from std_srvs.srv import Trigger
from rcl_interfaces.srv import ListParameters
from rclpy.callback_groups import MutuallyExclusiveCallbackGroup

class MyServer:

    def __init__(self, node):
        self._node = node

    def initialize(self):
        srv_name = '/rviz/list_parameters'
        param_client = self._node.create_client(ListParameters, srv_name)
        while not param_client.wait_for_service(timeout_sec=1.0):
            self._node.get_logger().info(f'Waiting for service {srv_name}...')
        req = ListParameters.Request()
        future = param_client.call_async(req)
        rclpy.spin_until_future_complete(self._node, future)  # ***
        self._node.get_logger().info("Initialized")

        self.launch_nav_srv_cb_group = MutuallyExclusiveCallbackGroup()
        self.launch_nav_srv = self._node.create_service(
            Trigger,
            '/my_service',
            self.my_callback,
            callback_group=self.launch_nav_srv_cb_group
        )

    def my_callback(self, request, response):
        self._node.get_logger().info('DONE')
        response.success = True
        response.message = 'Service completed successfully'
        return response

def main(args=None):
    rclpy.init(args=args)

    node = rclpy.create_node('spinning_node')
    executor = rclpy.executors.MultiThreadedExecutor()
    executor.add_node(node)

    spin_thread = threading.Thread(target=executor.spin, daemon=True)
    spin_thread.start()

    server = MyServer(node)
    server.initialize()

    spin_thread.join()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```
Here's the client code. I am pretty sure the client code is working as expected, since I've used it to call the AddTwoInts service from the documentation and that works.
```
# my_client.py
import rclpy
from rclpy.node import Node
from std_srvs.srv import Trigger

def main(args=None):
    rclpy.init(args=args)

    node = Node('minimal_client_node')
    client = node.create_client(Trigger, '/my_service')  # service name

    # Wait until the service is available
    while not client.wait_for_service(timeout_sec=1.0):
        node.get_logger().info('Waiting for service...')

    # Prepare the request
    node.get_logger().info('Making a request...')
    request = Trigger.Request()

    # Call the service asynchronously
    future = client.call_async(request)

    # Block until the future completes (service response or error)
    rclpy.spin_until_future_complete(node, future)

    if future.result() is not None:
        node.get_logger().info(f'Result: {future.result()}')
    else:
        node.get_logger().info('Service call failed')

    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

To reproduce the issue:
1. In one terminal, run `python minimal_buggy_server.py`
2. In another terminal, run `python my_client.py`
You will notice that `minimal_buggy_server.py` hangs after printing `Initialized`, and `my_client.py` hangs after printing `Making a request...`

3. Now, comment out  `rclpy.spin_until_future_complete(self._node, future)`, i.e. the line marked `# ***`
4. Repeat steps 1 and 2.
You will notice the service request gets received and callback gets executed; "DONE" will be printed on the server end.

#### Expected behavior
After running steps 1 and 2 with `rclpy.spin_until_future_complete(self._node, future)` uncommented,
 the client should still be able to successfully make a request.

#### Actual behavior
Refer to the description in step 4 of "To reproduce the issue" above.

#### Additional information

I got stuck on this for many hours. It doesn't make sense to me.

I curated this example but in my actual application, the service callback is supposed to use information returned from calling the ListParameters service.

Barry-Xu-2018, 2025-01-14:
It looks like you added a Node to be executed by two executors.

````python
    executor = rclpy.executors.MultiThreadedExecutor()
    executor.add_node(node)

    spin_thread = threading.Thread(target=executor.spin, daemon=True)
````

And

```python
rclpy.spin_until_future_complete(self._node, future)  # ***
````

zkytony, 2025-01-14:
What should I do instead?

zkytony, 2025-01-14:
Discovered that I should pass in the `executor` and the following two alternatives would work:

* `rclpy.spin_until_future_complete(self._node, future, executor=self._executor)`
* `self._executor.rclpy.spin_until_future_complete(future)`

Also, I could switch the orders of 
```python
    server = MyServer(node, executor)
    server.initialize()
```
and 
```python
    spin_thread = threading.Thread(target=executor.spin, daemon=True)
    spin_thread.start()
```
and it makes no difference. This seems to make sense as `spin_until_future_complete` would spin the node itself.

fujitatomoya, 2025-01-14:
the root cause for deadlock is already mentioned, `MultiThreadedExecutor` is not running any nodes after `rclpy.spin_until_future_complete(self._node, future)`.

> Discovered that I should pass in the executor and the following two alternatives would work:

i suggest this approach.

is there anything we need to discuss here? if not, can you close the issue?

zkytony, 2025-01-14:
Thanks for the clarification.

closed 2025-01-14 by zkytony