ros2/rclpy #1412: Where is rclpy docs/api for Humble?

scotgopal, 2025-02-12:
Hi there, good day! 

I can't seem to find the website for the hosted documentation for `rclpy` humble.

I found it for rolling: https://docs.ros.org/en/rolling/p/rclpy/rclpy.html

https://docs.ros.org/en/humble/p/rclpy/rclpy.html -> Returns 404 Not found

Thank you.

Yadunund, 2025-02-13:
Hi @scotgopal,

Following the release of Humble, we added support in our documentation generation tool, [rosdoc2]( https://github.com/ros-infrastructure/rosdoc2), to generate documentation for python packages and special packages like `rclpy` which is an `ament_cmake` package with python modules. This also required a few changes to rclpy as seen in these PRs
-  https://github.com/ros2/rclpy/pull/1103
- https://github.com/ros2/rclpy/pull/1109

Hence, documentation for rclpy is properly generated ROS 2 Iron Irwini onwards.

It should be possible to backport/adapt changes to the `humble` branch to get documentation generation working for Humble. Would you be willing to contribute the changes needed? We'd be happy to review a PR.

scotgopal, 2025-02-14:
Hi @Yadunund , thanks for engaging!

I will try to look into this on the weekend! I will also take some time to get familiar with  rosdoc2 and analyse what exactly needs to be backported to humble.

My plan here is to first try to use rosdoc2 to generate the docs locally, and then move from there. If I have any questions regarding the implementation, I will post here.

scotgopal, 2025-02-15:
Hi there.

I have made some preliminary changes in my own fork here (https://github.com/scotgopal/rclpy/tree/scotgopal/fix-humble-docs). 

Similar to https://github.com/ros2/rclpy/pull/1109, the docs seems to have generated well but with 349 warnings (4 more than the doc generation in rolling branch). I'll be happy to address any comments to amend these warnings during the PR review.

PR opened here (https://github.com/ros2/rclpy/pull/1414)

audrow, 2025-02-27:
I think this has been fixed.

PR #1414 (2025-02-23): Fix doc generation for Humble
Resolves #1412 

Fixes doc generation for Humble

backports of #1109 and #1103
closed 2025-02-27 by audrow