ros2/rclpy #1417: Client calls sometimes get stuck

firesurfer, 2025-02-19:
# Description

I have a setup where I am spinning a SingleThreaded Executor in python thread (simply by calling `rclpy.spin` and I am doing stuff in another thread.

We are now running into the issue that sometimes (usually after some hours of operation) one of our Clients get stuck.

#### Pseudo code of the client call

The usual call procedure looks like this

```
future = my_service_client.call_async(request)

while not future.done():
     time.sleep(0.001)
    # Additional code which raises a TimeoutError after x seconds (2s in the current case) 
```

The call procedure works fine like this for hours and thousands of service calls but then it just gets stuck and our process fails with a TimeoutError. 

As this just happens once in a while it is hard to reproduce.

# System

ROS2 Version: Jazzy
rlcpy: 7.1.3
Binary install

fujitatomoya, 2025-02-24:
could you provide the self-contained reproducible example including server side, unless that would be really hart to tell what is really going on here.

i would assume that,
- service server is fairly stable and responsive only for this client? what if it does not have `Additional code which raises a TimeoutError after x seconds (2s in the current case)`? it gets stuck forever within this sleep loop?
- when the problem happens, what happens if call the server service via `ros2 service call`? it responds back to the request or also stuck?

in the end, it would be much easier to have reproducible sample.

firesurfer, 2025-03-27:
Sorry for coming back so late to this issue. @fujitatomoya  

I spent some more time on this issue and it was actually a rather dumb mistake on our side. In some really really rare cases the service simply took longer to answer that what I specified as a timeout.

closed 2025-03-27 by firesurfer

fujitatomoya, 2025-03-27:
@firesurfer no worries, that happens to me a lot 😅  we really appreciate you for letting us know the update!