ros2/rclpy #1423: set_parameters callback returns None

belalhmedan90, 2025-03-05:
The callback returns None, and I get an error: `AttributeError; 'NoneType' object has no attribute 'successful'
in [this line 832](https://github.com/ros2/rclpy/blob/humble/rclpy/rclpy/node.py#L832)

fujitatomoya, 2025-03-05:
can you provide more description about this?

- what is the expected behavior?
- what is the actual behavior?
- self-contained, complete reproducible procedure or sample code.

belalhmedan90, 2025-03-06:
@fujitatomoya if the result of this [callback](https://github.com/ros2/rclpy/blob/humble/rclpy/rclpy/node.py#L831) is None, there should be an exception handling, or something like that imo.

InvincibleRMC, 2025-03-06:
Strange on Jazzy I am unable to reproduce.

```python
import rclpy
from rclpy.node import Node

class Test(Node):
    def __init__(self):
        super().__init__(node_name="test")
        params_list = [
            ("param1", "val1"),
            ("param2", "val2"),
            ("param3", "val3")
        ]
        self.declare_parameters(namespace="package_name", parameters=params_list)

rclpy.init()
a = Test()
print(a.get_parameters(["package_name.param1", "package_name.param2", "package_name.param3"]))
```

belalhmedan90, 2025-03-07:
> Strange on Jazzy I am unable to reproduce.
> 
> import rclpy
> from rclpy.node import Node
> 
> class Test(Node):
>     def __init__(self):
>         super().__init__(node_name="test")
>         params_list = [
>             ("param1", "val1"),
>             ("param2", "val2"),
>             ("param3", "val3")
>         ]
>         self.declare_parameters(namespace="package_name", parameters=params_list)
> 
> rclpy.init()
> a = Test()
> print(a.get_parameters(["package_name.param1", "package_name.param2", "package_name.param3"]))

True, the example itself doesn't generate this Error, but when combined with `ActionServer`, I will try to make a minimal reproducible example.

bhomaidan1990, 2025-03-08:
@fujitatomoya @InvincibleRMC  the issue is I forgot to add `return SetParametersResult(successful=True)` at the end of my parameters callback function, however I think better Exception to tell the user that his params callback is not returning anything might be more explicit, thanks.

fujitatomoya, 2025-03-08:
@bhomaidan1990 
CC: @InvincibleRMC @belalhmedan90 

thanks for checking on this 👍 

> I forgot to add return SetParametersResult(successful=True) at the end of my parameters callback function

yeah, that is the cause of this exception reported with this issue.

> however I think better Exception to tell the user that his params callback is not returning anything might be more explicit, thanks.

agree. `rclpy` should detect this and report it to the user application as system library. i can patch it.

fujitatomoya, 2025-03-08:
@bhomaidan1990 if you can review https://github.com/ros2/rclpy/pull/1425, that would be appreciated.

fujitatomoya, 2025-03-10:
https://github.com/ros2/rclpy/pull/1425 is not merged yet, so i just reopened this one.

closed 2025-03-10 by ahcorde