ros2/rclpy #1424: C++ code style linting possibly broken

alsora, 2025-03-08:
We noticed that a PR passes linting tests even if it has some inconsistent C++ style (e.g. mixture of `snake_case` and `PascalCase`).

See https://github.com/ros2/rclpy/pull/1391

We should review whether the linter is running as expected in this repository

bmartin427, 2025-03-11:
The [Google Style Guide](https://google.github.io/styleguide/cppguide.html#Function_Names) says:

> Regular functions have mixed case; accessors and mutators may be named like variables.

The changes in style in that PR are deliberate per this policy.  Not sure if ROS policy differs from this; I had read the [ROS policy](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Code-Style-Language-Versions.html#function-and-method-naming) as allowing snake_case more often, but not necessarily prohibiting CamelCase, but perhaps I was mistaken.

alsora, 2025-03-13:
Ok, so I think we can close this ticket? Unless we want to modify the ROS policy

mjcarroll, 2025-03-20:
> Unless we want to modify the ROS policy

Probably out of the scope of this ticket :)

closed 2025-03-20 by mjcarroll