ros2/rclpy #1452: Performance issues with ```MultiThreadedExecutor```

zacharyyamaoka, 2025-04-30:
Hi Rclpy Team,

Please see self-contained code example below which shows a simplified case of the issue. I am on Ubuntu Noble with Jazzy Distro

Here is a simple node that publishes to `/clock`

```python
#!/usr/bin/env python3

import rclpy
from rclpy.node import Node
from rosgraph_msgs.msg import Clock

class ClockPublisher(Node):
    def __init__(self):
        super().__init__('clock_publisher')
        self.publisher_ = self.create_publisher(Clock, '/clock', 10)
        publish_hz = 10
        self.timer = self.create_timer(1/publish_hz, self.timer_callback) 
        self.get_logger().info('Publishing rosgraph_msgs/Clock to /clock')

    def timer_callback(self):
        now = self.get_clock().now().to_msg()
        clock_msg = Clock(clock=now)
        self.publisher_.publish(clock_msg)
        self.get_logger().debug(f'Published Clock: {clock_msg.clock.sec}.{clock_msg.clock.nanosec:09d}')

def main(args=None):
    rclpy.init(args=args)
    node = ClockPublisher()
    try:
        rclpy.spin(node)
    except KeyboardInterrupt:
        pass
    finally:
        node.destroy_node()
        rclpy.shutdown()

if __name__ == '__main__':
    main()

```

Outputs
```bash
(venv) bam@bam-GPU:~/bam_ws/src$ ros2 topic hz /clock
average rate: 10.001
	min: 0.100s max: 0.100s std dev: 0.00016s window: 11
average rate: 10.001
	min: 0.100s max: 0.100s std dev: 0.00014s window: 22
average rate: 10.001
	min: 0.100s max: 0.100s std dev: 0.00014s window: 33
average rate: 10.001
	min: 0.100s max: 0.100s std dev: 0.00014s window: 43
average rate: 10.000
	min: 0.100s max: 0.100s std dev: 0.00013s window: 53
```
```bash
(venv) bam@bam-GPU:~/bam_ws/src$ ros2 topic echo /clock 
clock:
  sec: 1746021058
  nanosec: 425310669
---
clock:
  sec: 1746021058
  nanosec: 525296735
---
clock:
  sec: 1746021058
  nanosec: 625305596
[... 226 lines cut ...]

  rclcpp::shutdown();
  return 0;
}
```

Previously I had experinced this performance issue when making subscriptions to /camera/color (which I thought was also due to the large msg size) and /tf and /jointstates (which I thought was primarily due to the very high frequency). 

I also experinced it when using set `use_sim_time:=true` for a rclpy multithreaded node, and the performance became terrible. I realized this is beacuse it created a subscription to the /clock. 

I think there may be some more serious underlying issues as in this example, its unable to keep up with a simple /clock signal at 10hz.

--

I would be helpful to gets your thoughts on:

- What is the best practice for making subscriptions to topics from multi-threaded executors in rclpy?
- What are your thoughts using c++ buffer nodes and making service calls to deal with this issue? ( I got this idea after seeing the `tf2_buffer_client`)
- Longer term I imagine I could implement the nodes in c++, but I love rclpy for the pythonic similicity. Would using rclcpp fix the issues I am expericing right now? Is there a way to get better performance in rclpy?

Thank you for all your work on this excellent library! I ❤️ ROS2 :)

Best,

Zach

zacharyyamaoka, 2025-04-30:
I have also tested with different callbackgroup setups and see similar performance issues:

Timer and Subscription on their own `MutuallyExclusiveCallbackGroup`
```python
class ClockSubscriberWithTimer(Node):
    def __init__(self):
        super().__init__('clock_sub_with_timer')

        self.timer_cb_group = MutuallyExclusiveCallbackGroup()
        self.sub_cb_group = MutuallyExclusiveCallbackGroup()

        # Subscribe to /clock
        if True:

            self.subscription = self.create_subscription(
                Clock,
                '/clock',
                self.clock_callback,
                10,
                callback_group=self.sub_cb_group
            )

        # Timer
        timer_hz = 10
        self.timer_period = 1/timer_hz  # seconds
        self.timer = self.create_timer(self.timer_period, self.timer_callback, self.timer_cb_group)

```

Outputs

```bash
venv(venv) bam@bam-GPU:~/bam_ws/src$ /home/bam/bam_ws/venv/bin/python /home/bam/bam_ws/src/bam_dev/tests/multithreaded_node.py
[INFO] [1746023793.675794926] [clock_sub_with_timer]: Node started with multithreaded executor
[INFO] [1746023794.166265392] [clock_sub_with_timer]: [timer] freq avg: 9.99 Hz | period avg: 100.1 ms ± 0.3 ms
[INFO] [1746023795.468528407] [clock_sub_with_timer]: [clock] freq avg: 2.78 Hz | period avg: 359.4 ms ± 595.1 ms
[INFO] [1746023799.166462011] [clock_sub_with_timer]: [timer] freq avg: 5.88 Hz | period avg: 170.0 ms ± 148.5 ms
[INFO] [1746023801.175551123] [clock_sub_with_timer]: [clock] freq avg: 10.42 Hz | period avg: 96.0 ms ± 282.2 ms
[INFO] [1746023804.605975449] [clock_sub_with_timer]: [timer] freq avg: 3.02 Hz | period avg: 331.1 ms ± 328.8 ms
[INFO] [1746023806.762030761] [clock_sub_with_timer]: [clock] freq avg: 10.60 Hz | period avg: 94.3 ms ± 267.6 ms
[INFO] [1746023810.517243144] [clock_sub_with_timer]: [timer] freq avg: 2.70 Hz | period avg: 370.2 ms ± 373.2 ms
[INFO] [1746023811.826220689] [clock_sub_with_timer]: [clock] freq avg: 36.06 Hz | period avg: 27.7 ms ± 42.9 ms
```

Both on the same `MutuallyExclusiveCallbackGroup`

```python
class ClockSubscriberWithTimer(Node):
    def __init__(self):
        super().__init__('clock_sub_with_timer')

        self.sub_cb_group = MutuallyExclusiveCallbackGroup()

        # Subscribe to /clock
        if True:

            self.subscription = self.create_subscription(
                Clock,
                '/clock',
                self.clock_callback,
                10,
                callback_group=self.sub_cb_group
            )

        # Timer
        timer_hz = 10
        self.timer_period = 1/timer_hz  # seconds
        self.timer = self.create_timer(self.timer_period, self.timer_callback, self.sub_cb_group)
```

Outputs

```bash
venv(venv) bam@bam-GPU:~/bam_ws/src$ /home/bam/bam_ws/venv/bin/python /home/bam/bam_ws/src/bam_dev/tests/multithreaded_node.py
[INFO] [1746023901.542613853] [clock_sub_with_timer]: Node started with multithreaded executor
[INFO] [1746023902.157324745] [clock_sub_with_timer]: [clock] freq avg: 8.01 Hz | period avg: 124.9 ms ± 106.9 ms
[INFO] [1746023903.203789359] [clock_sub_with_timer]: [timer] freq avg: 2.99 Hz | period avg: 334.2 ms ± 368.3 ms
[INFO] [1746023907.694821573] [clock_sub_with_timer]: [clock] freq avg: 5.33 Hz | period avg: 187.5 ms ± 372.7 ms
[INFO] [1746023908.600694899] [clock_sub_with_timer]: [timer] freq avg: 1.85 Hz | period avg: 539.7 ms ± 368.2 ms
[INFO] [1746023912.710127770] [clock_sub_with_timer]: [clock] freq avg: 7.51 Hz | period avg: 133.1 ms ± 208.6 ms
[INFO] [1746023914.468160984] [clock_sub_with_timer]: [timer] freq avg: 3.24 Hz | period avg: 308.9 ms ± 306.0 ms
[INFO] [1746023918.186742892] [clock_sub_with_timer]: [clock] freq avg: 5.87 Hz | period avg: 170.2 ms ± 299.5 ms
[INFO] [1746023919.487122017] [clock_sub_with_timer]: [timer] freq avg: 2.48 Hz | period avg: 402.7 ms ± 346.2 ms
[INFO] [1746023923.448345058] [clock_sub_with_timer]: [clock] freq avg: 12.86 Hz | period avg: 77.8 ms ± 195.2 ms
```

Barry-Xu-2018, 2025-05-07:
FYI, maybe you can use the Events Executor for improved performance, which had been backported to Jazzy(https://github.com/ros2/rclpy/pull/1435).

zacharyyamaoka, 2025-05-10:
Thanks @Barry-Xu-2018! I did a bit of reserach into the Events Executor...

https://discourse.ros.org/t/faster-rclpy-executor-now-in-rolling/42852

Its not clear to me if this can support making service calls from within a service call though? Would that cause deadlock?

Thanks,

Zach

Barry-Xu-2018, 2025-05-11:
> Its not clear to me if this can support making service calls from within a service call though? Would that cause deadlock?

Although I haven't used this new executor much, I don't think it will cause any deadlock issues.

> Is there anywhere I can learn more about using this executor?

You can refer to test code https://github.com/ros2/rclpy/blob/3414456ddaf1163d1951d301336028c9600bd58e/rclpy/test/test_events_executor.py#L385-L393

zacharyyamaoka, 2025-05-13:
Amazing, this is a great reference. I will give it a go....