ros2/rclpy #1460: Declaring static type paramaters no longer matches rclcpp functionaility and always raises an exception

gstorer, 2025-05-15:
### Operating System:

Ubuntu 24

### ROS version or commit hash:

Jazzy

### Client library (if applicable):

rclpy

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

I have noticed this issue when upgrading our code base from Humble to Jazzy.

When declaring a static typed parameter in rclpy, it is no longer possible to supply a value a run time, a default value must be specified.

The following code runs fine in Humble but raises an exception in Jazzy when called as follows:
`python3 test_node.py --ros-args --param my_string_param:=test_value`
`import def __init__(self):
        super().__init__('simple_parameter_node')
        
        # Declare a string parameter without a default value using a descriptor to set its type
        string_descriptor = ParameterDescriptor(
            type=ParameterType.PARAMETER_STRING,
            description='A required string parameter'
        )
    
        self.my_param = self.declare_parameter(
            'my_string_param',
            descriptor=string_descriptor
        )
            
        # Print the parameter value
        self.get_logger().info(f'Received parameter: {repr(self.my_param.value)}')

def main(args=None):
    rclpy.init(args=args)
    node = SimpleParameterNode()
    
    try:
        rclpy.spin(node)
    except KeyboardInterrupt:
        pass
    finally:
        node.destroy_node()
        rclpy.shutdown()

if __name__ == '__main__':
    main()`

The exception raised is:
`ValueError: Cannot declare a statically typed parameter with default value of type PARAMETER_NOT_SET`

### Expected behavior

The behavior should match rclcpp. If a static typed parameter is declared without a default, it must be supplied when running the node or an exception is raised. 

### Actual behavior

rclpy always raises an exception for a static parameter without a default, even if it is supplied at runtime.

### Additional information

This appears to be the pull request were the issue was introduced: https://github.com/ros2/rclpy/pull/1216 . In it @haianos correctly identifies some user stories were it will be an issue. Note that before this pull request rclpy behavior didn't match rclcpp either, instead it never raised an exception. However, this didn't prevent my code from working.

For an example of the use of static typed parameters without defaults from within the ros2 org (in C++) see: https://github.com/ros2/openrobotics_darknet_ros/blob/6c741578d55b19ffb9a120b79cb8a78c7b60fbd6/src/detector_node.cpp#L98

gstorer, 2025-05-16:
Yes, the vast majority of our code is C++ and this is the pattern we exclusively use to ensure that configuration is only in one place.

This (admittedly slightly older) pull request makes it clear that static types without a default are intended to be a valid use:
https://github.com/ros2/rclcpp/pull/1522

The following C++ code demonstrates this and runs as I described:
```
#include "rclcpp/rclcpp.hpp"
#include "rcl_interfaces/msg/parameter_descriptor.hpp"
#include "rcl_interfaces/msg/parameter_type.hpp"

class SimpleParameterNode : public rclcpp::Node
{
public:
    SimpleParameterNode() : Node("simple_parameter_node")
    {
        rcl_interfaces::msg::ParameterDescriptor string_descriptor;
        string_descriptor.type = rcl_interfaces::msg::ParameterType::PARAMETER_STRING;
        string_descriptor.description = "A required string parameter";

        this->declare_parameter<std::string>("my_string_param", string_descriptor);

        std::string my_param = this->get_parameter("my_string_param").as_string();
        RCLCPP_INFO(this->get_logger(), "Received parameter: '%s'", my_param.c_str());
    }
};

int main(int argc, char **argv)
{
    rclcpp::init(argc, argv);
    rclcpp::spin(std::make_shared<SimpleParameterNode>());
    rclcpp::shutdown();
    return 0;
}
```

gstorer, 2025-05-21:
While digging around in rclpy to create a work around wrapper for my problem I came across this pull request that also suggests that the exception is intended to be raised when accessing an uninitialized variable, not on declaration: https://github.com/ros2/rclpy/pull/798

fujitatomoya, 2025-05-21:
@gstorer thanks for digging in this. i will look into this.