ros2/rclpy #1480: 🧑‍🌾 `test_executor_spin_until_future_complete_timeout` flaky failure in windows

Crola1702, 2025-07-16:
### Operating System:

Windows 10

### ROS version or commit hash:

Rolling

### Client library (if applicable):

rclpy

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

1. Run a build in windows (e.g., [windows repeated](https://ci.ros2.org/job/nightly_win_rep/3796/))
2. See test regressions failures

### Expected behavior

Tests should pass

### Actual behavior

Tests are failing

### Additional information

Reference builds:
- [nightly_win_rel#3480](https://ci.ros2.org/job/nightly_win_rel/3480)
- [nightly_win_rep#3796](https://ci.ros2.org/job/nightly_win_rep/3796)

Test regressions:
[projectroot.test_executor](https://ci.ros2.org/job/nightly_win_rep/3796/testReport/junit/(root)/projectroot/test_executor/)
[rclpy.rclpy.test.test_executor.TestExecutor.test_executor_spin_until_future_complete_timeout](https://ci.ros2.org/job/nightly_win_rep/3796/testReport/junit/rclpy.rclpy.test.test_executor/TestExecutor/test_executor_spin_until_future_complete_timeout/)

Log output:

```
self = <test.test_executor.TestExecutor testMethod=test_executor_spin_until_future_complete_timeout>

    def test_executor_spin_until_future_complete_timeout(self) -> None:
        self.assertIsNotNone(self.node.handle)
        for cls in [SingleThreadedExecutor, EventsExecutor]:
            with self.subTest(cls=cls):
                executor = cls(context=self.context)
                executor.add_node(self.node)
    
                def timer_callback() -> None:
                    pass
                timer = self.node.create_timer(0.003, timer_callback)
    
                # Timeout
                future = Future[None]()
                self.assertFalse(future.done())
                start = time.perf_counter()
                executor.spin_until_future_complete(future=future, timeout_sec=0.1)
                end = time.perf_counter()
                # Nothing is ever setting the future, so this should have waited
                # at least 0.1 seconds.
>               self.assertGreaterEqual(end - start, 0.1)
E               AssertionError: 0.0979279987514019 not greater than or equal to 0.1

..\..\src\ros2\rclpy\rclpy\test\test_executor.py:500: AssertionError
- generated xml file: C:\ci\ws\build\rclpy\test_results\rclpy\test_executor.xunit.xml
```

This test started failing in 03/22/2025, after #1391. See [changes](https://github.com/ros2/rclpy/pull/1391/files#diff-a8a69522e9ab0729b33d099b87442b8a0bad35f42f46097558a4b22d1cced174L448-L479)

Flakiness:

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_rep|2025-07-15|2025-07-05|9|7|77.78
nightly_win_rel|2025-07-06|2025-07-06|11|1|9.09

miguelgonrod, 2026-08-19:
Updating flakyness info

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
|--|--|--|--|--|--|
nightly_win_rep|2026-08-16|2026-08-06|4|4|100.0
nightly_win_rel|2026-08-16|2026-08-05|8|2|25.0

fujitatomoya, 2026-08-20:
@bmartin427 can you take a look at the quick patch https://github.com/ros2/rclpy/pull/1712? what do you think?