ros2/rclpy #1484: `rclpy` crashes on exit in tests

hidmic, 2025-07-31:
### Operating System:

```Linux 43219d45bc38 6.14.0-24-generic #24~24.04.3-Ubuntu SMP PREEMPT_DYNAMIC Mon Jul  7 16:39:17 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

rolling

### Client library (if applicable):

rclpy

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : UP,LOOPBACK,RUNNING
mtu          : 65536
inet         : 172.17.0.2
inet4        : ['172.17.0.2']
ether        : 92:2a:51:6b:1a:f1
netmask      : 255.255.0.0
device       : eth0
flags        : UP,BROADCAST,RUNNING,MULTICAST
mtu          : 1500
broadcast    : 172.17.255.255

   PACKAGE VERSIONS
launch_testing_ros                        : latest=0.29.2, local=0.29.2
ament_cmake_ros                           : latest=0.15.2, local=0.15.2
rmw_test_fixture_implementation           : latest=0.15.2, local=0.15.2
launch_ros                                : latest=0.29.2, local=0.29.2
rclpy                                     : latest=10.0.0, local=10.0.0
rcl_lifecycle                             : latest=10.2.3, local=10.2.3
rcl_action                                : latest=10.2.3, local=10.2.3
rcl                                       : latest=10.2.3, local=10.2.3
rmw_implementation                        : latest=3.1.2, local=3.1.2
rmw_zenoh_cpp                             : latest=0.8.2, local=0.8.2
zenoh_cpp_vendor                          : latest=0.8.2, local=0.8.2
test_msgs                                 : latest=2.4.2, local=2.4.2
std_msgs                                  : latest=5.8.2, local=5.8.2
rosgraph_msgs                             : latest=2.4.2, local=2.4.2
rmw_fastrtps_dynamic_cpp                  : latest=9.4.2, local=9.4.2
rmw_fastrtps_cpp                          : latest=9.4.2, local=9.4.2
rmw_fastrtps_shared_cpp                   : latest=9.4.2, local=9.4.2
rmw_cyclonedds_cpp                        : latest=4.1.1, local=4.1.1
rmw_connextddsmicro                       : latest=N/A, local=1.2.2
rmw_connextdds                            : latest=1.2.2, local=1.2.2
rmw_connextdds_common                     : latest=1.2.2, local=1.2.2
rmw_dds_common                            : latest=4.0.0, local=4.0.0
composition_interfaces                    : latest=2.4.2, local=2.4.2
rcl_interfaces                            : latest=2.4.2, local=2.4.2
lifecycle_msgs                            : latest=2.4.2, local=2.4.2
rosidl_default_runtime                    : latest=1.8.1, local=1.8.1
rosidl_default_generators                 : latest=1.8.1, local=1.8.1
action_msgs                               : latest=2.4.2, local=2.4.2
unique_identifier_msgs                    : latest=2.8.1, local=2.8.1
type_description_interfaces               : latest=2.4.2, local=2.4.2
service_msgs                              : latest=2.4.2, local=2.4.2
builtin_interfaces                        : latest=2.4.2, local=2.4.2
rosidl_core_runtime                       : latest=0.4.0, local=0.4.0
rosidl_core_generators                    : latest=0.4.0, local=0.4.0
[... 395 lines cut ...]
    at Python/bytecodes.c:2706                                                                                                                                                                                                                                                                                        
#29 0x0000000000555136 in _PyEval_EvalFrame (throwflag=0, frame=0x7f51800140a8, tstate=<optimized out>) at ../Include/internal/pycore_ceval.h:89                                                                                                                                                                      
#30 gen_send_ex2 (closing=0, exc=0, presult=<synthetic pointer>, arg=0x0, gen=0x7f5180014060) at ../Objects/genobject.c:230                                 
--Type <RET> for more, q to quit, c to continue without paging--                                                                                                                                                                                                                                                      
#31 gen_iternext (gen=0x7f5180014060) at ../Objects/genobject.c:604                                                                                                                                                                                                                                                   
#32 0x00000000005d3d4c in builtin_next_impl (module=<optimized out>, default_value=<optimized out>, iterator=<generator at remote 0x7f5180014060>) at ../Include/object.h:220                                                                                                                                         
#33 builtin_next (module=<optimized out>, args=0x7f51b6c163e0, nargs=<optimized out>) at ../Python/clinic/bltinmodule.c.h:730                               
#34 0x00000000005d80cb in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at ../Include/cpython/methodobject.h:50                                                                                                                                                 
#35 0x000000000054cb32 in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=1, args=0x7f518f756bf8, callable=<function at remote 0x7f51b4296020>, tstate=0x7f51982d4f30)
    at ../Include/internal/pycore_call.h:92                                             
#36 method_vectorcall (method=<optimized out>, args=0xb49140 <_PyRuntime+76288>, nargsf=<optimized out>, kwnames=0x0) at ../Objects/classobject.c:69
#37 0x00000000005dad16 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/bytecodes.c:3254
#38 0x000000000054cb32 in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=1, args=0x7f518f756e18, callable=<function at remote 0x7f51b8da2d40>, tstate=0x7f51982d4f30)
    at ../Include/internal/pycore_call.h:92                                                                                                                                      
#39 method_vectorcall (method=<optimized out>, args=0xb49140 <_PyRuntime+76288>, nargsf=<optimized out>, kwnames=0x0) at ../Objects/classobject.c:69                                                                                                    
#40 0x00000000006f80dc in thread_run (boot_raw=0x7f51982d48d0) at ../Modules/_threadmodule.c:1114                                                                                                                                                       
#41 0x00000000006b931c in pythread_wrapper (arg=<optimized out>) at ../Python/thread_pthread.h:237                                                                                                                                                      
#42 0x00007f51b9044aa4 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:447                                                                                                                                                             
#43 0x00007f51b90d1a34 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100                    

```

</details>

Something's off with GIL management in `rclpy` bindings.

Blast545, 2025-07-31:
🧑‍🌾 Taking a look at this atm

Blast545, 2025-08-01:
🧑‍🌾 Not sure what's the long term solution, but as a temporary measure, this should fix it:
https://github.com/eProsima/Fast-DDS/pull/5968

Blast545, 2025-08-01:
Should be addressed after https://github.com/eProsima/Fast-DDS/pull/5968 and https://github.com/eProsima/Fast-DDS/pull/5973, let us know if you still find this problem, otherwise we'll close this issue.

mjcarroll, 2025-12-10:
Attempted fix here: https://github.com/ros2/launch_ros/pull/511