ros2/rclpy #1489: Cannot shutdown context from callbacks

fennel-labs, 2025-08-09:
### Operating System:

Docker:  ros:humble-perception-jammy (Ubuntu 22)

### ROS version or commit hash:

Docker:  ros:humble-perception-jammy (Humble)

### Client library (if applicable):

rclpy

### 'ros2 doctor --report' output

### Steps to reproduce issue

Execute the following node code:

````python
import rclpy
from rclpy.node import Node

class ShutdownTimerNode(Node):
    def __init__(self):
        super().__init__('shutdown_timer_node')
        self.create_timer(2.0, self.timer_callback)

    def timer_callback(self):
        self.get_logger().info('Timer triggered, shutting down...')
        rclpy.shutdown()

def main(args=None):
    rclpy.init(args=args)
    node = ShutdownTimerNode()
    try:
        node.get_logger().info('Node is running, waiting for timer...')
        rclpy.spin(node)
    except KeyboardInterrupt:
        node.get_logger().info('Keyboard interrupt received, shutting down...')

if __name__ == '__main__':
    main()
````

### Expected behavior

According to the [documentation ](https://docs.ros.org/en/humble/p/rclpy/api/init_shutdown.html), `rclpy.shutdown()` shuts down the node context including the executor. `rclpy.spin()` blocks until the executor is shut down. Thus, calling `rclpy.shutdown()` from the timer callback should cause the node to terminate. 

### Actual behavior

The node does not terminate. This means either the `rclpy` has a bug or the documentation is incorrect. If the documentation is incorrect, I would like to learn about the intended way to shutdown a ROS node without using `sys.exit()`.

christophebedard, 2025-08-09:
Can you first try with Kilted (or ideally Rolling) to see if the behaviour is different?

fennel-labs, 2025-08-09:
Rolling does not exhibit the observed behavior.

MichaelOrlov, 2025-08-14:
Hi @fennel-labs. At ROS 2 weekly triage meeting, we came to the conclusion that yes, that is a bug that is fixed in further ROS 2 distro releases. 
However, the community help wanted here to figure out what PR fixed this bug, so that it can be considered for backporting to the Humble.